rg-engineering/ioBroker.sbfspot #9: Error im Log bei jeder Ausführung, trotzdem funktional

mcmuller, 2018-03-12:
Hallo,
ich habe bei jeder Ausführung u.a. Fehler (Error: Cannot enqueue Quit after invoking quit.)  im iobroker-Log. Der Adapter tut aber, was er soll. Gibt's da eine Erklärung für? 
Grüße!

`

host.spok.local | 2018-03-12 19:15:00.351 | info | instance system.adapter.sbfspot.0 terminated with code 0 (OK)
-- | -- | -- | --
sbfspot.0 | 2018-03-12 19:15:00.345 | info | terminating
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Query.EofPacket (/opt/iobroker/node_modules/mysql/lib/protocol/sequences/Query.js:123:8)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Query._handleFinalResultPacket (/opt/iobroker/node_modules/mysql/lib/protocol/sequences/Query.js:139:8)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Query.Sequence.end (/opt/iobroker/node_modules/mysql/lib/protocol/sequences/Sequence.js:88:24)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Query._callback (/opt/iobroker/node_modules/iobroker.sbfspot/sbfspot.js:856:13)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at CalcHistory_Months (/opt/iobroker/node_modules/iobroker.sbfspot/sbfspot.js:887:9)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at DB_Disconnect (/opt/iobroker/node_modules/iobroker.sbfspot/sbfspot.js:899:26)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Connection.end (/opt/iobroker/node_modules/mysql/lib/Connection.js:249:18)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Protocol.quit (/opt/iobroker/node_modules/mysql/lib/protocol/Protocol.js:92:23)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Protocol._enqueue (/opt/iobroker/node_modules/mysql/lib/protocol/Protocol.js:139:13)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | at Protocol._validateEnqueue (/opt/iobroker/node_modules/mysql/lib/protocol/Protocol.js:204:16)
sbfspot.0 | 2018-03-12 19:15:00.342 | error | Error: Cannot enqueue Quit after invoking quit.
sbfspot.0 | 2018-03-12 19:15:00.342 | error | uncaught exception: Cannot enqueue Quit after invoking quit.
sbfspot.0 | 2018-03-12 19:15:00.278 | info | got data from STP 10000TL-10 with ID 2110068155
sbfspot.0 | 2018-03-12 19:15:00.276 | info | got data from Bluetooth Repeater with ID 170004554
sbfspot.0 | 2018-03-12 19:15:00.219 | info | start with mySQL
sbfspot.0 | 2018-03-12 19:15:00.213 | info | starting. Version 2.2.0 in /opt/iobroker/node_modules/iobroker.sbfspot, node: v8.10.0
host.spok.local | 2018-03-12 19:15:00.010 | info | instance system.adapter.sbfspot.0 started with pid 5228

`

rg-engineering, 2018-03-14:
Hallo @mcmuller, kann ich auch ein log mit debug - Infos bekommen, welches den Fehler zeigt? Danke.

mcmuller, 2018-03-14:
..klar doch:

[sbfspot-iobroker-debug.log](https://github.com/rg-engineering/ioBroker.sbfspot/files/1811108/sbfspot-iobroker-debug.log)

rg-engineering, 2018-03-15:
Das Problem war das Schliessen der Datenbank, wenn mehrere Wechselrichter abgefragt werden. Mit den letzten beiden commits (V2.2.1) sollte das Problem behoben sein...

mcmuller, 2018-03-15:
Danke, das wars!

closed 2018-03-15 by mcmuller