rg-engineering/ioBroker.sbfspot #11: Display in SBFspot Widget no as expected

Tottback, 2019-01-28:
SBFspot widget showing not the expected result with sqlite3 database
With "today" no values appears. The others are working but no x-axes appear.
Some values in years/12 months also appear strange.

[image]
[image]
[image]
[image]
[image]

Tottback, 2019-01-28:
X-axis now appear after changing the colour (!= 0xffffff). The todays view still show no values.

rg-engineering, 2020-04-05:
solved with 2.4.3

closed 2020-04-05 by rg-engineering