rg-engineering/ioBroker.sbfspot #17: SMA Wechselrichter SPR-4600M-DE wird nicht erkannt

MadMe86, 2019-06-22:
Hallo!
Ich versuche seit geraumer Zeit nun die PV-Analge meines Vaters in iobroker einzubinden...
Leider ziemlich erfolglos.
Der Aufbau ist folgender(Anlage von Sunpower):
Der Wechselrichter: SPR-4600M-A
ist verbunden mit einem Datenlogger: spr-pmr-dlog
und dies ist dann mittels Funk mit spr-pmr-gtwy verbunden.
Das Gateway sendet die Daten an Sunpower. Die konnte ich auch mitsniffen und kriege auch sinnvolle Daten raus...
Das Script sbfspot-config gibt als Fehler leider nur "ERROR: Connection to inverter failed!" aus.
Ich gehe mal davon aus das die Anlage noch nicht unterstützt wird...
Irgendwelche Tipps?

rg-engineering, 2019-06-22:
Zunächst musst du sbfspot zum laufen bringen. Tipps gibt es hier: https://github.com/SBFspot/SBFspot.
Laut sbfspot benötigst du eine bluetooth oder Ethernet Verbindung zwischen dem Wechselrichter und dem system, wo sbfspot läuft.
Der Adapter hier greift nicht direkt auf den Wechselrichter zu. Der Adapter liest nur die Daten aus der Datenbank, welche sbfspot schreibt.

MadMe86, 2019-06-22:
Oh man da hab ich was vermischt^^
Danke!

closed 2019-06-22 by MadMe86