rg-engineering/ioBroker.sbfspot #18: unable to open database file

martinschm, 2019-06-29:
Hi,
ich bekomme die Meldung, das die Datenbank nicht geöffnet werden kann. Installation von SBFspot ist als root erfolgt, iobroker über Installationsskript.

Wenn ich per Shell die Datenbank abfrage, bekomme ich Daten.
Die Datei liegt unter /home/smadata/SBFspot.db

`sbfspot.0 | 2019-06-29 20:30:07.970 | warn | force terminate in connect
-- | -- | -- | --
sbfspot.0 | 2019-06-29 20:30:02.018 | error | Error while performing Query in GetInverter. Error: SQLITE_CANTOPEN: unable to open database file
sbfspot.0 | 2019-06-29 20:30:01.969 | debug | SELECT * from Inverters
sbfspot.0 | 2019-06-29 20:30:01.968 | debug | sqlite Database is connected ...
sbfspot.0 | 2019-06-29 20:30:01.962 | debug | --- connecting to /home/smadata/SBFspot.db
`

rg-engineering, 2019-06-30:
laut https://forum.iobroker.net/topic/5555/neuer-adapter-sma-wechselrichter-mit-sbfspot/73 benötigt der user iobroker Lese- und Schreibrechte auf die Datenbank-Date bzw. den Ordner, wo die Datenbank-Datei liegt...

martinschm, 2019-06-30:
Hi, ja hab es da im Forum gepostet. Hab die Lösung durch experimentieren raus gefunden.

closed 2019-06-30 by martinschm