rg-engineering/ioBroker.sbfspot #38: iobroker object tree structure for sbfspot.0 with empty value fields

sebastianlinz, 2020-07-26:
In iobroker's object tree no values are set for object sbfspot.0. All value fields are empty.

Setup is used with sqlite. After adapter startup object tree structure for sbfspot.0 is created. 

Log shows different select statements and result values on debug level (eg. serial number, type, total). Database access seems to work. Log ends with:

sbfspot.0 | 2020-07-26 16:45:01.512 | info | (32335) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
-- | -- | -- | --
sbfspot.0 | 2020-07-26 16:45:01.511 | info | (32335) all done ...
sbfspot.0 | 2020-07-26 16:45:01.510 | debug | (32335) disconnect database

But object tree structure has empty value fields. 

Trying to access sbfspot object values from blockly's object chooser was not successfull. sbfspot adapter is not proposed.

rg-engineering, 2020-07-26:
can you please show me a complete log (probably with deletion of serial number and data base query results if you want)

sebastianlinz, 2020-07-26:
see part of log file:
[iobroker.2020-07-26.log](https://github.com/rg-engineering/ioBroker.sbfspot/files/4977912/iobroker.2020-07-26.log)

sebastianlinz, 2020-07-27:
Could the issue be similiar to this: https://github.com/ioBroker/ioBroker.javascript/issues/405? There it says values should be "JSON.stringyfied".

sebastianlinz, 2020-07-27:
I played around adding log statements. For example in main.js at line 962 total produced energy is available. But there seems to be an error writing state. Possibly it's not related to adapter's method "setState()". I tried it with simple pair "id" and "value" with no result. 

Maybe there's problem with the adapter object and my configuration. My iobroker is quite new, only a few days. Reading "https://github.com/ioBroker/ioBroker/wiki/Adapter-Development-Documentation#how-to-create-own-adapter" I found that creation of adapters seems to be changed. At least this could explain why sbspot adapter is not accessible from blocklys adapter chooser.

Hope this helps :-)

rg-engineering, 2020-07-27:
can you please show me your object tree where you expect your data? should look like this:
[image]

is your serial number shown as parent channel?
Do you run other adapters on the same system?

your log looks good, there is no problem...

sebastianlinz, 2020-07-27:
this is how object tree looks like: 
[image]

yes, serial number is shown as parent channel id and name.

yes, I'm succefully running other adapters: eg. influxdb, smartmeter, sonoff, telegram

sebastianlinz, 2020-07-27:
Object element "TotalPac" has a name. There is a difference in main.js line 207

rg-engineering, 2020-07-27:
can you please delete sbfspot.0 and all children? and then run the adapter again?
after that can you please click on the edit pen of any state and post raw data (right hand side tab)
should look like this:
[image]

> There is a difference in main.js line 207

you mean the additional name in  adapter.extendObject? I added this to rename the name in existing installations (which works well here)
The problem here is that adapter.setObjectNotExists does not create a complete and valid datapoint...

rg-engineering, 2020-07-27:
forget the previous post: I see the same probelm with a new clean installation.... need to fix it...

rg-engineering, 2020-07-29:
@sebastianlinz please install latest version here from github. After installation version number should be 4.0.0 Please delete complete data tree from sbfspot.0 before...

closed 2020-09-21 by rg-engineering