rg-engineering/ioBroker.sbfspot #276: exception in DB_GetInverters (exception in DB_GetInverters [SqliteError: attempt to write a readonly database])

runrennerrun, 2024-02-05:
Hi, 

although I see data from my inverter coming in via mqtt (not as I like but at least it's coming in) I get the following errors in the log. I'm not sure what to do to fix them. 

[image]

many thanks in advance!

rg-engineering, 2024-02-05:
Please check the access rights of your folder, where you placed your sqlite database file. 

see also here:
https://forum.iobroker.net/topic/5555/neuer-adapter-sma-wechselrichter-mit-sbfspot/229

runrennerrun, 2024-02-05:
okay, thanks for the quick reply. With which command I do this best? sudo chmod -R 775 . ?

runrennerrun, 2024-02-20:
> okay, thanks for the quick reply. With which command I do this best? sudo chmod -R 775 . ?

sudo chown iobroker SBFspot.db

closed 2024-02-20 by runrennerrun