rg-engineering/ioBroker.sbfspot #421: Update in Iobroker fails with this error:

holgerpommerien, 2024-09-28:
$ iobroker upgrade sbfspot@4.3.4

Update sbfspot from @4.3.3 to @4.3.4

host.raspberrypi Adapter "system.adapter.sbfspot.0" is stopped.

NPM version: 10.2.3Installing iobroker.sbfspot@4.3.4... (System call)

npm ERR! code 1npm ERR! path /opt/iobroker/node_modules/better-sqlite3npm ERR! command failednpm ERR! command sh -c prebuild-install || node-gyp rebuild --releasenpm ERR! make: Entering directory '/opt/iobroker/node_modules/better-sqlite3/build'npm ERR! TOUCH ba23eeee118cd63e16015df367567cb043fed872.intermediatenpm ERR! ACTION deps_sqlite3_gyp_locate_sqlite3_target_copy_builtin_sqlite3 ba23eeee118cd63e16015df367567cb043fed872.intermediatenpm ERR! TOUCH Release/obj.target/deps/locate_sqlite3.stampnpm ERR! CC(target) Release/obj.target/sqlite3/gen/sqlite3/sqlite3.onpm ERR! rm -f Release/obj.target/deps/sqlite3.a Release/obj.target/deps/sqlite3.a.ar-file-list; mkdir -p dirname Release/obj.target/deps/sqlite3.anpm ERR! ar crs Release/obj.target/deps/sqlite3.a @Release/obj.target/deps/sqlite3.a.ar-file-listnpm ERR! COPY Release/sqlite3.anpm ERR! CXX(target) Release/obj.target/better_sqlite3/src/better_sqlite3.onpm ERR! rm ba23eeee118cd63e16015df367567cb043fed872.intermediatenpm ERR! make: Leaving directory '/opt/iobroker/node_modules/better-sqlite3/build'npm ERR! g++: error: unrecognized command line option ‘-std=c++20’; did you mean ‘-std=c++2a’?npm ERR! make: *** [better_sqlite3.target.mk:123: Release/obj.target/better_sqlite3/src/better_sqlite3.o] Error 1npm ERR! gyp ERR! build error npm ERR! gyp ERR! stack Error: make failed with exit code: 2npm ERR! gyp ERR! stack at ChildProcess. (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)npm ERR! gyp ERR! System Linux 5.10.17-v7l+npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "--release"npm ERR! gyp ERR! cwd /opt/iobroker/node_modules/better-sqlite3npm ERR! gyp ERR! node -v v18.19.0npm ERR! gyp ERR! node-gyp -v v10.0.1npm ERR! gyp ERR! not oknpm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-09-26T19_46_11_212Z-debug-0.log

host.raspberrypi Cannot install iobroker.sbfspot@4.3.4: 1

ERROR: Process exited with code 25

rg-engineering, 2024-09-28:
Which version of the operating system are you using? I guess it's a Linux on Raspi?

holgerpommerien, 2024-10-02:
Linux raspberrypi 5.10.17-v7l+ #1421

On a Raspi 4